jflight-public/jesc #1: EFM8BB1 Targets

AlessandroAU, 2019-07-24:
Is there a hardware limitation regarding having this work on EFM8BB1 targets?

Does it fit in memory? I feel like it would be very useful even at reduced dshot rates, particularly for micros and whoops.

joelucid, 2019-08-04:
It would likely fit in memory. The biggest problem is the lack of timer4 on efm8bb1. If it weren't for that at least dshot-300 would likely work. Not impossible to work around but there aren't really many efm8bb1 boards still around.

AlessandroAU, 2019-08-04:
Most whoop boards are efm8bb1, I would argue that whoops would stand to benefit the most from reduced filter latency. Since the tlm is sent between dshot packets and timer 0 is only used during dshot reception why can it be used to bitbang the tlm too?

joelucid, 2019-08-04:
> Since the tlm is sent between dshot packets and timer 0 is only used during dshot reception why can't it be used to bitbang the tlm too?

Mostly historic reasons and to have a clear separation between the hw features used by JESC vs Telemetry. There is a performance benefit to using a dedicated timer for the bit banging though.

As I said it's not impossible to do - just fairly significant effort. Depends a bit on user feedback, too, whether it's worth doing.

joelucid, 2020-01-24:
Efm8bb1 targets are now available.

closed 2020-01-24 by joelucid