jflight-public/jesc #6: Motor wires de-soldering on Nameless AIO 412(T)

stylesuxx, 2019-09-17:
I came across a very strange issue - unfortunately not personally, but I have now heard from two people that the solder joints of the motors get so hot after flashing JESC that the motor wires de-solder. Re-flashing the original version of BLHELI_S does not seem to solve the issue.

One of those cases (and attempt of debugging) is documented in the comments here => https://brushlesswhoop.com/blheli_s-rpm-filter/ NLRC 412T, 1206 motors on 3S, BF4.1 RC1.

Another case I came across on Facebook just now - NLRC 412T, 1304 motors on 4S.

Interestingly enough I have no problems on my NLRC AIO 412 with 1103 motors on 2S, BF4,1 RC1.

I would like to help debugging, but I am not sure which information I can help out with.

Ericarias84, 2019-09-17:
I am the guy with the 1304 motors.  Very odd issue.  I am running this firmware on a bunch of other builds without issues (some AIO fcs).  After I flashed the firmware, the motors initially armed but than I got some smoke from the board and it appears some of the escs burned out.

etracer65, 2019-09-17:
@stylesuxx 

> Re-flashing the original version of BLHELI_S does not seem to solve the issue.

That would seem to rule out the ESC firmware don't you think?

@Ericarias84 

Sometimes motors or ESCs just fail.

Make sure you've set the `motor_poles` parameter in Betaflight correctly. Little tiny motors like that may not be the default 14 poles. If you don't have this set properly the RPM Filter won't work properly and it's possible that the motors will get hot. But it wouldn't cause a motor to smoke instantly.

stylesuxx, 2019-09-17:
@etracer65 the first case was working before flashing JESC and would still overheat even after being reverted to it's original state (BF and BLHELI_S). The `motor_poles` were set correctly to 12.

etracer65, 2019-09-17:
> and would still overheat even after being reverted to it's original state (BF and BLHELI_S)

That's my point. This would rule out a firmware problem and point to a hardware problem.

stylesuxx, 2019-09-17:
OK, so you are saying it was just a coincidence that the problems started **after** flashing JESC?

Wrong firmware version was also my first guess but we double checked that afterwards.

Unfortunately I can only provide second hand information - just wanted to post here to make people aware of a potential problem with this constellation. Since it happened two times already, chances are that it might happen again.

Thanks for your input.

stylesuxx, 2019-09-18:
So I just spoke to NamelessRC. One of their engineers looked into it and said that you should use ESC version P-H-90. Both boards come flashed with P-H-15.

So @etracer65 was right with his assumption that the manufacturer shipped the board with the wrong settings.

Find attached a screenshot of my conversation with NamelessRC on facebook.

[image]

Ericarias84, 2019-09-18:
That is interesting.  How did you and a few other people get it to work on your Nameless board?  Which version do you have flashed?

stylesuxx, 2019-09-18:
From what I understand, only people with 2S got it working without issues. I flashed P-H-15 of JESC, it came with P-H-15 of BLHELI_S. As it seems I was just lucky with my 2S setup not to run into any problems.

I have heard from another guy, also running 2S but DSHOT600 that he was getting temp warnings in beta. This might be an indicator too, that it is running too hot even on 2S.

etracer65, 2019-09-19:
@stylesuxx The page that's apparently related to you at: https://brushlesswhoop.com/namelessrc-nlrc-aio-412t-jesc/ is very misleading.

The only news here is that the manufacturer shipped the ESCs with the wrong firmware settings. The settings they supply were incorrect enough to cause physical damage to the ESCs. The deadtime is derived based on the electrical characteristics for the FETs based on the manufacturer design and datasheet. It is not something that would be set differently for different firmwares or something you would "tune". The manufacturer just screwed up. If I owned one of these ESCs I would be contacting the manufacturer for replacements even if they haven't failed yet because they're already likely damaged to some extent.

Is it possible that people flying with RPM filtering enabled using JESC might be working the ESC harder due to better flight characteristics? Sure. Could this put more stress on the already marginal ESC? Sure. But that's not the underlying cause. The incorrect deadtime is.

joelucid, 2019-09-19:
Hey guys - great you got this issue sorted out. Not sure why the issue started when installing Jesc. Jesc uses the very same code as blhelis for managing pwm in 24khz pwm mode. The experimental 48khz mode has some differences but none that could cause this. 

Beyond coincidence the only explanation I can come up with is that the manufacturer installed a firmware that was tagged as P-H-15 but was really a P-H-90. This would cause exactly the issues described here: it would work as purchased but Jesc would install the P-H-15 version and even after flashing back to blhelis you would now have a version that doesn’t work. Any update in blhelis configurator would have led to the same outcome in that case 

Etracer is also right: it’s entirely possible that the esc went bad over time and just started to show symptoms by coincidence when Jesc was installed.

stylesuxx, 2019-09-19:
@joelucid I was also suspecting that it might simply be tagged wrongly, but in the meantime NamelessRC replied and stated that no - they in fact do flash **P-H-15** in the factory:
[image]

I now asked him why this would be necessary in their opinion. Is there any way I can (easily) dump the current firmware from the firmware to compare the hex files.

@etracer65 thank you for your feedback regarding my article. I now added a disclaimer that people should not flash different versions in attempts of "tuning" their quadcopters.

As you can see from the above screenshot the manufacturer states that **P-H-15** is indeed the correct firmware that is flashed in the factory and that we should use **P-H-90** if we want to use JESC. I also did ask him about his reasoning for this and that people here are of the opinion that a different deadtime for JESC makes no sense.

I also invited their engineer to join the discussion here.

joelucid, 2019-09-19:
@Ericarias84, did you use the experimental 48khz mode or 24khz?

joelucid, 2019-09-19:
I found an issue that the experimental 48khz pwm mode scaled the dead time down in half. This needs to be done for any power related variables since the pwm runs twice as fast. But not for the dead time.

@Ericarias84, if you did in fact use the experimental 48khz pwm version that could explain the issue. This would also explain why some people don't have a problem (the ones on the stable 24khz pwm version) and others do (users of the 48khz pwm beta).

joelucid, 2019-09-19:
I pulled the 48khz beta for now until I have the time to address the issue.

joelucid, 2019-09-19:
Just flew 10 packs through the fixed 48khz pwm Version and had no issues. Would anybody like to test this out in the nameless aio? I can provide a hex.

stylesuxx, 2019-09-19:
@joelucid the first guy that contacted me, flashed JESC according to my screens - which show 24kHz (which is also the default selection).
[image]

I'd try it, but the Problem does not show on my 2S build anyways, so I am not of much help :-/

joelucid, 2019-09-19:
@stylesuxx - ok. Well it still seems he's unsure so who knows. Many people are in the bigger is better dept - so maybe it was an unconscious choice. I've flown the 24khz version so much that I'm sure there's nothing similar in there.

Ericarias84, 2019-09-19:
I also used 24 when I flashed.

joelucid, 2019-09-19:
Well - that brings us back to the hypothesis that P-H-15 might not be a suitable fw for the esc. 24khz pwm mode has been flown on hundreds of quads without this issue. It does seem to be hw specific then.

stylesuxx, 2019-09-19:
@joelucid OK - so which options do we have to confirm this hypothesis?
* Dump the initial firmware from the ESC and compare to the **P-H-15** HEX - I assume this is not easily possible?
* Someone with a known working 3S setup without JESC has to re-flash the ESC's with BLHELI_S **P-H-15**. If his setup is still working as before we can assume that indeed this version is flashed in the factory. If not, we know they flashed the wrong version. If it is still working the next step would be to flash JESC of the same version and confirm that it fries the ESC's.

Any other - possibly non destructive options? Maybe I should try and measure the temperature of the FET's on the bench with 2S @ 20% throttle: let the motors spin for 20s, measure the temp of the FET's with JESC, flash BLHELI_S and  re-do the measurement. Not sure how accurate the measurement would be - what do you think?

joelucid, 2019-09-19:
>  Maybe I should try and measure the temperature of the FET's on the bench with 2S @ 20% throttle: let the motors spin for 20s, measure the temp of the FET's with JESC, flash BLHELI_S and re-do the measurement. Not sure how accurate the measurement would be - what do you think?

Yeah that's a good idea. You don't even need to connect motors. The dead time prevents the upper and lower side FETs to short the power supply while switching - so it's independent of any external load. This makes it very easy to spot: no load, just a bit of throttle in the motors tab. If the dead time setting is ok the fets should not get warm. If the fets get warm you should have enough time to pull the motor back to zero to prevent any damage.

If it kills the ESC on 3s it should be enough for the ESCs to get warm on 2s. You can then also check if there's any difference if you use P-H-90.

stylesuxx, 2019-09-19:
OK, I will try the following constellations and see if the FETs get hot:

| Variant | Hot |
|----------|------|
| BF 4.0, BLHELI_S - STOCK | |
| BF 4.1, BLEHLI_S - STOCK | |
| BF 4.0, BLHELI_S - flashed P-H-15 | |
| BF 4.1, BLHELI_S - flashed P-H-15 | |
| BF 4.0, BLHELI_S - flashed P-H-90 | |
| BF 4.1, BLHELI_S - flashed P-H-90 | |
| BF 4.1 JESC - P-H-15 | |
| BF 4.1 JESC - P-H-90 | |

Did I miss anything?

joelucid, 2019-09-20:
@stylesuxx sounds like a plan! There's probably not much of a need to test BF 4.0 though.

stylesuxx, 2019-09-20:
@joelucid Yes, I thought that too, that's why I only tested with BF 4.1 in different constellations - I edited my initial post already btw.

joelucid, 2019-09-20:
Sorry didn’t see you already had results. Well this certainly is very fascinating. I’ll try to reproduce. Thanks for testing!

joelucid, 2019-09-20:
So far I can't reproduce this with my debug escs. Anybody know how to get a hold of a Nameless AIO 412(T) in Germany?

stylesuxx, 2019-09-20:
@joelucid I'll send you mine - I am from AT. Message me: stylesuxx@gmail.com
But I would like to have it back if at all possible.

joelucid, 2019-09-20:
I have prepared one test we could still try: could you flash the Telemetry hex called Test version (internal use only) using jesc configurator? I've disabled actually sending telemetry but this still works with the inverted dshot protocol. That way we can eliminate that as factor.

stylesuxx, 2019-09-20:
@joelucid stays at 40C with this telemetry hex.

joelucid, 2019-09-20:
One more idea: go back to the normal Telemetry hex but set your loop time to 1k.

stylesuxx, 2019-09-20:
Gets slightly less hot than with 4k.

Ericarias84, 2019-09-20:
Question for you guys not related to this problem.  I have a 6s build running 32bit escs.  I am running RPM filter firmware on the escs, is it safe to use 48 PWM frequency or should I stick with 24?

RadianceNL, 2019-10-15:
Is there any news on this? I just received my board and would like to flash JESC with telemetry but obvious don't want to fry my board.

joelucid, 2019-10-15:
I’ll be back from vacation tomorrow. Hopefully Chris board will have arrived so I can run some tests.

RadianceNL, 2019-10-15:
Thanks for the quick reply. I'll be patient. Weather here is crap anyways. :-)

joelucid, 2019-10-16:
Alright, the board arrived here and I did some first tests. I flashed different FW's and held my thumb on he escs while increasing throttle to about 1150 in the motors tab. I used a 2s battery. With P-H-15 the FETs became hot very quickly in all cases (BLHeli_S, JESC + Telemetry + Bidir on and JESC + Telemetry + Bidir off). With P-H-90 the FETs didn't get hot in any case at all.

Personally I couldn't observe a huge difference between the different tests that did get hot. I'm inclined to think that the FETs just need a higher dead time than 15. Unfortunately the FETs don't seem to be marked so I can't look them up in a datasheet.

I'll try to contact nameless to get the FET specs.

joelucid, 2019-10-16:
BTW even P-H-30 was already much, much better. Probably not necessary to go to P-H-90.

stylesuxx, 2019-10-16:
Interesting - that contradicts my tests. And brings us technically back to square one: Why were people able to fly it without issues before flashing but not after flashing JESC? So all of the people that smoked their FET's after flashing JESC had just bed luck, or do the FET's have to work harder after flashing JESC and having RPM filtering enabled?

Thank you for looking into this and thank you for contacting nameless. I always hit them up on facebook and they tend to respond rather quickly.

joelucid, 2019-10-16:
Yeah @stylesuxx it is strange. One thing that's different in jesc is that I only apply the new rc packet to pwm once the telemetry packet is sent. That could influence timing a bit.

The good thing about what I found is that it is consistent with the other data - we haven't seen this issue on any other esc. Hopefully nameless will provide the FET specs so we can confirm what timings should be consistent with the switching speed of the FETs.

stylesuxx, 2019-10-16:
I hope so too - but since it would probably mean, that they have to admit a mistake I am wondering if it will happen.

Did you test if there is any difference in temp with P-H-30 when using 3 or 4S?

joelucid, 2019-10-16:
So far I only tested 2s. No response from nameless on FB thus far.

joelucid, 2019-10-17:
Ok - new insights: tried blhelis 16.7 P-H-15 on 3s and the fets get too hot to touch immediately when applying power.

I also asked around and  @conuthead provided this input:
> @Joe Lucid  Is that a passive design? ( no gate driver) . If it is the circuit I think it is .. it uses p channel upper fets and is super slow to turn off. I use the _90 dead time.

Looking at the board it does indeed look like there are no gate drivers:

[image]

Finally nameless answered this morning and said they wouldn't share the FET specs. They also gave this advice:
[image]

Which kind of sounds like not any P-H-15 fw will do - it has to be the one they preinstalled. I probed further but they went silent.

Unless I hear anything else from nameless I think anybody updating their FW on this esc should use P-H-90 as they recommend for jesc. And that includes updates to BLHeli_S FW.

joelucid, 2019-10-17:
Here's the top of the board. It has 3 transistors per esc. That makes it likely this design (https://www.rcgroups.com/forums/showthread.php?2898397-Another-strange-BlHeli_S-1-3s-DIY-esc) which is the one that requires a dead-time of 90.

Maybe there is an incentive to make ESCs appear to switch faster than they actually can for marketing purposes.

[image]

stylesuxx, 2019-10-17:
Great detective work - thank you. Interestingly enough they do not post ESC timings anywhere on their product pages. But them telling us not to upgrade BLHEL_S at all really sounds strange.

I guess our only way to find out the truth would be dumping the ESC FW from a new board and compare that to a known P-H-15. I guess that's not worth anyone's time, better get the info out to only use the P-H-90 FW no matter with BLHELI_S or JESC.

joelucid, 2019-10-17:
Yeah - if this sort of thing happens more often I would need to implement some kind of checksumming of the installed fw to avoid upgrading mistagged versions. But so far it’s just this one. Getting the word out would be good. I’ll add it to the install instructions. Maybe you could update your readership as well. I’ll keep the board for a couple of days in case anything comes back from nameless if that’s ok with you.

stylesuxx, 2019-10-17:
Will do - I will update [this article](https://brushlesswhoop.com/namelessrc-nlrc-aio-412t-jesc/) with your new findings. Sure, keep it another week or two.

joelucid, 2019-10-17:
@RadianceNL which platform do you run jesc-configurator on? I could make a version that extracts the firmware.

RadianceNL, 2019-10-17:
Windows 10

RadianceNL, 2019-10-17:
[Log.txt](https://github.com/jflight-public/jesc/files/3740329/Log.txt)
There you go. :-) Didn't take that long so hopes it's oke.

joelucid, 2019-10-17:
I'm sorry, @RadianceNL, I have to ask you to run it once again. Every other page is missing including just the one with the timing. https://jflight.net/downloads/JESC-Configurator_win64_1.2.4b.zip is the fixed version.

RadianceNL, 2019-10-17:
[Log_2.txt](https://github.com/jflight-public/jesc/files/3740617/Log_2.txt)
No problem at all!

RadianceNL, 2019-10-17:
I'm glad I could help a bit so no problem.
I scrolled through the log a bit but that's all Abra Kadabra to me.

But conclusion to this? Run 90 like Nameless RC suggested in the first place? Or are you going to make a 120 version like they did and should I wait for that?

joelucid, 2019-10-17:
Just run 90 that should be fine for now. I’ve contacted them again - hopefully we can come to a joint solution.

RadianceNL, 2019-10-17:
Just did and did some test hoovers inside the house and all works fine on a 3s 450mah lipo. Probably won't be able to fly it till Tuesday because the weather will stay crap until then.

Thanks for your time and effort looking into this!

stylesuxx, 2019-10-17:
OK, I don't know what to say - would be interested in a statement by NamelessRC.

joelucid, 2019-10-18:
Ok - did get some sign of life from namelessrc: 
[image]

That's it. I'm speechlessrc.

stylesuxx, 2019-10-18:
Hmm, maybe there is a language barrier? I was thinking about contacting Bob Roogie (KebabFPV) since he was involved in board layout, maybe he can get some more information - pretty sure he is also interested in shedding some light on this situation.

stylesuxx, 2019-10-18:
I told Bob about this thread, lets see if he chimes in here.

Sounds like a good reason for higher dead time - still I wish they would have been open about it - at least tag it with some value that indicates that there is something off.

I guess before RPM filtering was a thing chances were slim for anyone to update BLHELI_S on one of those AIO boards...

If the resistor is the weak link - is it an option that on the fried boards it would be enough to swap that resistor to get them working again, or is the chance too high, that something else blew too?

joelucid, 2019-10-18:
> If the resistor is the weak link - is it an option that on the fried boards it would be enough to swap that resistor to get them working again, or is the chance too high, that something else blew too?

Nah - it's weak in the sense that its heat dissipation limits how low the resistance can be. And the lower the resistance the faster the switching. So larger component allows better heat dissipation allows lower resistance.

But if the board needs 90-120 and you use 15 it will fry the escs rather quickly. Of course in some cases like the original poster you might just need to flash P-H-90 and be good again. That's if the FETs have not been fried yet.

stylesuxx, 2019-10-18:
I see, thanks for the explanation. And the big fat warning is a great idea. Thanks for all that time you put into this issue.

[5 more replies]

closed 2019-11-15 by joelucid