jflight-public/jesc #7: ESCs can't arm and don't beep after flashing JESC (works OK reflashing to regular blheli_s)

straferr, 2019-10-17:
I have these Aikon SEFM 30A ESCs running C-H-15 that fall completely silent after flashing to JESC (latest version configurator as at 16 Oct 2019).  They also won't arm.  This is totally independent of the FC and FC firmware.  When power is connected to ESC+Motor without FC, I don't even get the first usual beeps when power is applied.

The situation is reversible by reflashing with regular BLHeli_S.  It is also repeatable:  I have alternated between JESC and regular BLHeli_S at least 3 times now to test this.

The JESC firmware and telemetry flashing was done with all suggested/allowed defaults and there appeared to be no errors during the flashing process (log attached).

[Log.txt](https://github.com/jflight-public/jesc/files/3738452/Log.txt)

This is the same issue I first raised on RC Groups: [https://www.rcgroups.com/forums/showpost.php?p=42965907&postcount=233](url)

joelucid, 2019-10-17:
Ok - looking deeper this esc has a special flash page that jesc needs to modify locked. Unless you erase and flash it via the c2 interface there's no chance to get jesc on it.

joelucid, 2019-10-17:
Great to see that the workaround for this case - to run the boot loader - works and allows to flash blheli_s again.

If you're up for c2 flashing just flash blheli_s 16.7 after complete chip erase (see https://www.youtube.com/watch?v=b-7VFt3IhhE). Then a jesc install should work.

joelucid, 2019-10-17:
Reached out to Aikon to find out why the page is locked.

straferr, 2019-10-18:
Again, fantastic and super-speedy sleuthing work, joelucid! *thumbsup*  

Good to know there's a workaround, and I've already got a couple of Arduino's set up as flashing sticks, but I won't be able to get to the C2 pads on my Aikon escs at the moment.... they're under a thick coat of conformal coating, heatshrink and buried deep in my quad...lol.  I have another set of Aikons still in their baggies.  Will give it a try with those when the weekend finally lands (ie. soon :D).  Will update here once I've done this.  Thanks again!

jlandegent, 2019-10-19:
think i have the same issue on my esc's. beerotor 80A v2. when flash jesc firmware everything is fine, esc boots up, will respond in in motor control. as soon telemetry is flashed esc keeps silent.

straferr, 2019-10-21:
Just wanted to update on the suggested C2 flash workaround... TLDR; it works!  JESC is now working properly on my ESCs.

If anyone else is reading, this is what I had to do:

- locate the C2 pins on ESC and solder on at least the clock and data wires for a C2 flash

- use BLHelisuite to create a 4-way-interface using an Arduino Uno (use pins D12 and D13 for C2)

- flash regular blheli_s via c2 using the Arduino and blhelisuite (Note:  blhelisuite initially reported unknown firmware when I first hooked up the esc, and after that, it also complained about the esc being locked and will not proceed to flash... it even throws up some unfriendly error message.  But if you persist, subsequently it let me flash anyway.)

- Once regular blheli_s is on the esc, JESC flashes without issue.  You can then access the unit using JESC configurator as you would normally do :)

straferr, 2019-10-22:
I guess I'll close this issue since it is not actually a fault related to JESC per se :)

closed 2019-10-22 by straferr