jflight-public/jesc #11: the fc do not  reconize esc#1

pizkuy, 2020-01-21:
[Log.txt](https://github.com/jflight-public/jesc/files/4091757/Log.txt)

joelucid, 2020-01-24:
This is a very interesting log: while flashing ESC 3 jesc-configurator starts writing settings to ESC 1. This causes ESC 1 to get selected and it then receives part of the flash for ESC 3.

This should obviously not happen. Do you remember if you tried writing settings while flashing was in progress?

sending cmd_DeviceWrite 1900 255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,2,28,0
received cmd_DeviceWrite ACK_OK 1900 0
sending cmd_DeviceRead 1800 128
received cmd_DeviceRead ACK_OK 1800 255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255
sending cmd_DeviceRead 1880 128
received cmd_DeviceRead ACK_OK 1880 255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255
sending cmd_DeviceRead 1900 128
received cmd_DeviceRead ACK_OK 1900 255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255
sending cmd_DeviceRead 1980 128
**Writing setup
sending cmd_DeviceInitFlash 0 0**
received cmd_DeviceRead ACK_OK 1980 255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,2,28,0
sending cmd_DevicePageErase 0 9
received cmd_DeviceInitFlash ACK_OK 0 178,232,100,1
received cmd_DevicePageErase ACK_OK 1200 9
sending cmd_DeviceWrite 1200 255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255,255
received cmd_DeviceWrite ACK_OK 1200 0

joelucid, 2020-01-24:
I think I know what happened: the "Defaults" button is not disabled during flashing. Is it possible that you pressed it while flashing? I can reproduce flashing issues by pressing "Defaults" while a flash is in progress.

pizkuy, 2020-01-24:
I think yes, My computer freeze in the moment i press flashing than maybe
I press the debug button, is a possibility to rescue my esc board ?

pizkuy, 2020-01-27:
Can i rescue the esc ? 
-------- Mensaje original --------De: joelucid <notifications@github.com> Fecha: 24/1/20  00:30  (GMT-06:00) A: jflight-public/jesc <jesc@noreply.github.com> CC: pizkuy <pizkuy3@gmail.com>, Author <author@noreply.github.com> Asunto: Re: [jflight-public/jesc] the fc do not  reconize esc#1 (#11) I think I know what happened: the "Defaults" button is not disabled during flashing. Is it possible that you pressed it while flashing? I can reproduce flashing issues by pressing "Defaults" while a flash is in progress.

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/jflight-public/jesc/issues/11?email_source=notifications\u0026email_token=AOK4BIJBHESIZNLVNO5UKJLQ7KDIJA5CNFSM4KJVLVU2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEJZ3NWQ#issuecomment-578008794",
"url": "https://github.com/jflight-public/jesc/issues/11?email_source=notifications\u0026email_token=AOK4BIJBHESIZNLVNO5UKJLQ7KDIJA5CNFSM4KJVLVU2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEJZ3NWQ#issuecomment-578008794",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

joelucid, 2020-02-13:
You can always reflash an esc using the c2 interface. JB has a video on it: https://www.youtube.com/watch?v=b-7VFt3IhhE. I've now changed jesc-configurator to no longer allow pressing the "defaults" button during flashing.

closed 2020-02-13 by joelucid