jflight-public/jesc #20: iFlite Cinebee 75HD Hybrid 4K - 3 ESCs flash, last one will not

leeph, 2020-04-25:
I have bought a total of 8 licenses so far and have not had any success flashing JESC on any BLHeli_S ESC that I own, so far, and no ability to get the licenses back it seems?

I tried flashing JESC on my Cinebee as per subject, this has an iFlite SucceX 16x16 stack with BLHeli_S 4-in-1.

I am able to flash the first 3 ESCs, but the license button is greyed out so I can't enable RPM telemetry on them anyway. The 4th ESC fails to flash, even with ignore layout checked.

I have attached the debug log. Please assist, because I feel like I've just wasted money at this stage. Thank you in advance.
[jesc_flash_error.txt](https://github.com/jflight-public/jesc/files/4533435/jesc_flash_error.txt)

leeph, 2020-04-25:
Just as a bit of extra info, I've left the FC in this stack at Betaflight 3.5.7 (stock firmware) and was trying to update the ESCs first. Perhaps if someone could let me know if having Betaflight 4.1.x flashed first is a requirement for flashing the JESC firmware with telemetry that I want, some trouble could be saved.

If it is a requirement to flash BF4.1.x first, it would be useful for this to be made clear on the JESC FAQ page.

ntoenis, 2020-05-11:
Hello I also have this stack and I'm having the same issue. Esc 1-3 flash as expected and this is my second stack attempting to install jesc on. First one flashed all 4 fine but this second stack is having issues. Since I'm able to flash all four in blheli configurator as expected I assumed this was a jesc configurator issue. https://github.com/jflight-public/jesc-configurator/issues/59
[Log.txt](https://github.com/jflight-public/jesc/files/4611564/Log.txt)
[BlheliLogWorked.txt](https://github.com/jflight-public/jesc/files/4611565/BlheliLogWorked.txt)

ntoenis, 2020-05-11:
Do you also have one esc stating it's licensed even though you haven't licensed it yet? 
[image]
[image]

leeph, 2020-05-11:
Yes, I get that as well.

I've so far purchased 8 licenses of JESC and haven't managed to get it successfully flashed on all 4 ESCs on either quad I bought it for. And no contact from the developers. They've taken the money and offered nothing in the way of support.

Pretty terrible customer service, tbh. Not impressed or happy.

Resorted to using JazzMaverick's free code which seems to work just as good, and is free.

leeph, 2020-05-11:
16 days and the only response is from another unsatisfied user.... lol !!

ntoenis, 2020-05-11:
Must say the one stack it did function on was great, so thinking just some sort of issue a odd bit on these 4-in-1's or variations in the boards manufacturing suppliers resulting in slightly different builds. Thankfully the price isn't too bad but yeah that super sucks for 8*cost.. Are you also G-H-30?

leeph, 2020-05-11:
Yes, G-H-30.

I've read that the ESC may be locked by the manufacturer. But JESC doesn't warn you this may be the case prior to purchase, and gives you no ability to de-license them if it doesn't install, and no ability to refund if it doesn't work.

I've submitted two PayPal claims - fed up of being taken for a ride by people not willing to stand by their own product.

joelucid, 2020-05-11:
Hey guys - sorry for the late response. Both your logs show some escs which have a bootloader that has the signature of a L esc though all escs are really H escs. This is a mistake by the manufacturer and trips up jesc-configurator. I have had this issue before and did a workaround though I never released it yet.

joelucid, 2020-05-11:
I'll do a build and upload so you can see if it fixes your issues.

ntoenis, 2020-05-11:
OMG Senpai! I'm sorry but thank you so much for the response. Super appreciated

leeph, 2020-05-11:
That would be appreciated. I will hold off on the PayPal claim if you can produce a working... workaround.

joelucid, 2020-05-11:
Bits are uploading. I'm at the weekend cottage with crappy internet so it's taking a while. Macos, win32 and win64 enough for everybody?

ntoenis, 2020-05-11:
That works for me also super kind to be doing that while in the woods. Stay healthy friend (win64 :D)

leeph, 2020-05-11:
Win32 or 64 good for me.

Is this a custom version of the configurator only, or the firmware G-H-30 as well? I'm sorry I didn't fully understand your explanation, so please forgive my ignorance.

joelucid, 2020-05-11:
It's just a new configurator. This new version doesn't use the bootloader signature to determine whether it's an L or H esc but looks at the layout instead.

joelucid, 2020-05-11:
Ok try this: https://jflight.net/downloads/jesc-configurator-signature-workaround/

ntoenis, 2020-05-11:
Sick it opened up to the screen I expected with all unlicensed. Flashed all four successful and telemetry flashing working too(though I went with 2.2 and 2.1 for the initial test
[NewLogWorked.txt](https://github.com/jflight-public/jesc/files/4612299/NewLogWorked.txt)
). Log attached for the curious. Hella appreciated

joelucid, 2020-05-11:
Glad to hear @ntoenis. @leeph I hope this fixes your issue, too. Sorry for not looking at this earlier - just had a lot of other stuff going on.

leeph, 2020-05-12:
Thanks @joelucid that's got it.

Please can you let me know what the correct DShot settings are for this? I am referring specifically to dshot_bitbang and dshot_burst and a couple of other associated parameters.

I was under the impression that the bitbang setting would work if set to AUTO but the ESCs do not initialise with that setting enabled. I'm just keen to set it up 'the way it's supposed to be' - perhaps you could provide a config snippet for BF 4.1.6 ?

Thanks - I will close the PayPal case for this set of 4 licenses. The other set of 4 I purchased are still surplus to my requirements now as after the trouble flashing them, I replaced with BLHeli32 ESCs so I no longer have need for the other licenses.